tfyoung/esphome-junctek_kgf #1: TypeError when compiling

ckloeti, 2022-06-12:
Hello tfyoung, many thanks for programming this esphome sensor integration for Junctek battery monitor. A I try to compile it it gives me following error:
INFO Reading configuration /config/esphome/yucon-battery-monitor.yaml...
ERROR Unable to load component junctek_kgf.sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/9c6cd628/components/junctek_kgf/sensor.py", line 54, in <module>
    cv.Optional(CONF_VOLTAGE): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 5 were given
Failed config

sensor.junctek_kgf: [source /config/esphome/yucon-battery-monitor.yaml:18]
  
  Platform not found: 'sensor.junctek_kgf'.
  platform: junctek_kgf
  address: 1
  voltage: 
    name: Battery Voltage
  current: 
    name: Battery Current
  battery_level: 
    name: Battery Level
  temperature: 
    name: Battery Temperature

Are you able to help me as my programming skills are not so good to fix this issue.

Many thanks
Christoph

ckloeti, 2022-06-12:
Hi, I changed sensor.py to be working now. Just change CONFIG_SCHEMA to this

CONFIG_SCHEMA = cv.All(
    cv.Schema(
        {
            cv.GenerateID(): cv.declare_id(JuncTekKGF),
            cv.Optional(CONF_ADDRESS, default=1): cv.int_range(1, 255),
            cv.Optional(CONF_VOLTAGE): sensor.sensor_schema(
                unit_of_measurement=UNIT_VOLT,
                icon=ICON_FLASH,
                accuracy_decimals=1,
                device_class=DEVICE_CLASS_VOLTAGE,
                state_class=STATE_CLASS_MEASUREMENT,
            ),
            cv.Optional(CONF_CURRENT): sensor.sensor_schema(
                unit_of_measurement=UNIT_AMPERE,
                icon="mdi:current-dc",
                accuracy_decimals=1,
                device_class=DEVICE_CLASS_CURRENT,
                state_class=STATE_CLASS_MEASUREMENT,
            ),
            cv.Optional(CONF_BATTERY_LEVEL): sensor.sensor_schema(
                unit_of_measurement=UNIT_PERCENT,
                icon=ICON_PERCENT,
                accuracy_decimals=1,
                device_class=DEVICE_CLASS_BATTERY,
                state_class=STATE_CLASS_MEASUREMENT,
            ),
             cv.Optional(CONF_TEMPERATURE): sensor.sensor_schema(
                unit_of_measurement=UNIT_CELSIUS,
                icon=ICON_THERMOMETER,
                accuracy_decimals=0,
                device_class=DEVICE_CLASS_TEMPERATURE,
                state_class=STATE_CLASS_MEASUREMENT,
            ),
        }
    ).extend(uart.UART_DEVICE_SCHEMA)
    )

tfyoung, 2022-06-15:
Thanks for reporting this. I've been meaning to fix this having had the issue come up on another integration i'm working on (I haven't updtated my junktek for a while to hit this for this module). I'll push a fix for it soon, though if you wanted to take your changes and make a pull request for them, I'd gladly merge.

tfyoung, 2022-06-16:
This should be fixed now with the most recent commit to main

closed 2022-06-16 by tfyoung

ckloeti, 2022-06-17:
Many thanks tfyoung