tfyoung/esphome-junctek_kgf #2: Current polarity inverted

dann2008, 2022-10-11:
[image]

[image]

The values shown in HA are inverted but on the junktec display polarity is shown correctly

The green side is for Charging + and the red side is discharging -

How can i change polarity ?

Quattrohead, 2022-12-08:
Hi Dann, are you on diysolarforum ? Would like to chat to you about this on here or there.

dann2008, 2023-01-17:
Ok, let me now how are your readings 
I had to invert the shunt in order to display correct polarity
But its wrong, because now the battery capacity is inverted
Thank you

tfyoung, 2023-02-06:
Hi,

I'm still not sure if it's currently right or wrong (my setup has no load connected again yet to test), but I've added the option to change it. there is now an "invert_current" option. It defaults to "false" to match previous behaviour, but once i have things setup and confirm which way i think it should be, then that may change. @dann2008 you defintely want to add the "invert_current: true" line, to get the behaviour you want.

closed 2023-02-06 by tfyoung

dann2008, 2023-02-10:
[image]
Hi, 

I've just added the  "invert_current: true" line on my config, now the polarity is shown correctly
Values shown in the photo are positive but my battery is discharging, when i'll invert back the shunt it will show the correct values
Thank You !!!