tfyoung/esphome-junctek_kgf #4: Compiling issue

BogNik72, 2022-12-15:
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/api_connection.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/api_server.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/list_entities.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/proto.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/api/user_services.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/esp32/gpio.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/junctek_kgf/junctek_kgf.cpp.o
Compiling /data/batteria-ext/.pioenvs/batteria-ext/src/esphome/components/logger/logger.cpp.o
src/esphome/components/junctek_kgf/junctek_kgf.cpp: In function 'int getval(const char*&)':
src/esphome/components/junctek_kgf/junctek_kgf.cpp:19:46: error: exception handling disabled, use -fexceptions to enable
     throw std::runtime_error("invalid number");
                                              ^
In file included from src/esphome/components/sensor/sensor.h:3,
                 from src/esphome/components/junctek_kgf/junctek_kgf.h:4,
                 from src/esphome/components/junctek_kgf/junctek_kgf.cpp:1:
src/esphome/components/junctek_kgf/junctek_kgf.cpp: In member function 'void JuncTekKGF::handle_line()':
src/esphome/components/junctek_kgf/junctek_kgf.cpp:125:42: error: 'e' was not declared in this scope
     ESP_LOGE("JunkTekKGF", "Error %s %s",e.what(), buffer);
                                          ^
src/esphome/core/log.h:123:89: note: in definition of macro 'esph_log_e'
   esp_log_printf_(ESPHOME_LOG_LEVEL_ERROR, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                         ^~~~~~~~~~~
src/esphome/components/junctek_kgf/junctek_kgf.cpp:125:5: note: in expansion of macro 'ESP_LOGE'
     ESP_LOGE("JunkTekKGF", "Error %s %s",e.what(), buffer);

tronron, 2023-01-05:
I found exactly the same errors. Were you ever able to resolve this?

tfyoung, 2023-01-05:
First of all thanks for reporting this, secondly sorry for my delayed response. My unit is offline for various reasons and I haven't updated for a while, hoping to get it back online soon. Something seems to have changed with esphome to do with exceptions such that the code doesn't compile anymore. I'll try and get a fix in the next few days.

tronron, 2023-01-05:
Thank you! It looks like a great and very useful integration!Get Outlook for AndroidFrom: tfyoung ***@***.***>Sent: Thursday, January 5, 2023, 1:41 PMTo: tfyoung/esphome-junctek_kgf ***@***.***>Cc: tronron ***@***.***>; Comment ***@***.***>Subject: Re: [tfyoung/esphome-junctek_kgf] Compiling issue  (Issue #4)
First of all thanks for reporting this, secondly sorry for my delayed response. My unit is offline for various reasons and I haven't updated for a while, hoping to get it back online soon. Something seems to have changed with esphome to do with exceptions such that the code doesn't compile anymore. I'll try and get a fix in the next few days.

thecivvie, 2023-01-05:
Thanks, appreciate it. I was mentioning it on the Renogy forum

tfyoung, 2023-01-06:
I think I have fixed it in a test branch. As I said, my unit isn't currently hooked up, so all I have tested is that it compiles. The changes were to remove exceptions for the error path, since the latest esphome no longer supports them.

If you want you can test it by using 
```
external_components:
- source: github://tfyoung/esphome-junctek_kgf@usesetjmpfix
```
I'll try to get around to hooking it back up to my battery in the next few days and will push the fix to main once i've tested it.

thecivvie, 2023-01-06:
Compiled and linked fine. Now to find a spare esp32

tronron, 2023-01-06:
Hi,

 

I can confirm that it complies and links fine now also.

 

I have a few ESP-01s on hand so I may try with them over the weekend, but I think Tx, RX are 3.3v, which probably needs level shifted to work. I do have several of the Peacefaire PZEM AC units working with that chip. I supplied 3.3v into the 5v input and put another 1k resistor in parallel with one of the I/O opto couplers, as is described in HA help. I bet the I/O coupling is exactly the same in these units.

 

Otherwise, I’ll get a ESP 32 ASAP.

 

Thanks for your rapid response and help. I think this Junctek product is really good value, and with sales increasing I can see your integration being very popular!

 

Ron

tronron, 2023-01-08:
Hi - found an ESP 32, an RS 485-TTL convertor and connected everything up. Complies, loads and goes on-line perfectly. However, in the logs I only get:
[image].

I can see periodic (~1-10s ) communication on both the Tx and Rx lines, no other errors except"[10:06:31][E][JunkTekKGF:020]: Error invalid number"

shiftybugger, 2023-01-14:
> Hi - found an ESP 32, an RS 485-TTL convertor and connected everything up. Complies, loads and goes on-line perfectly. However, in the logs I only get: [image].
> 
> I can see periodic (~1-10s ) communication on both the Tx and Rx lines, no other errors except"[10:06:31][E][JunkTekKGF:020]: Error invalid number". With my scope I can decode the R50 request coming from the ESP, and see the requested data stream successfully entering the ESP 32, so it looks like the issue is with HA processing the data.

I can confirm tronron's findings. The test branch compiles fine for me and I get the same log results. I'm not a great coder but I'll see what I can see.

tronron, 2023-01-15:
Thank you shiftybugger, I really want to get this working, but I'm a hardware guy, not a coder at all. Please post any progress you might make!

thecivvie, 2023-01-17:
I compiled this ok, but I am not really into esp devices so not sure what I am doing wrong but getting no data from the UART. I must searched for a wirign diagram

shiftybugger, 2023-01-19:
I'm declaring defeat. I'm not familiar enough with C++ to unwind this.

It looks like  getval() inevitably ends up with (end==pos) evaluating to TRUE, jumping back to handle_line() and causing an immediate return. However, I can't figure out how to fix it. I'll try again if I get some time in the next week or so, but I'm a bit lost atm.

shiftybugger, 2023-01-30:
Defeat was short lived. I've submitted a pull request with my changes, which basically involve removing the error handling and replacing with some basic manual checking in logic. I also removed the calls to verify_checksum as I haven't gotten to reworking the logic in that to remove the reliance on try/catch.

Feel free to test my branch, shiftyfix. You can do that by changing the external component in your ESPHome yaml to:

```
external_components:
  source:
    type: git
    url: https://github.com/shiftybugger/esphome-junctek_kgf.git
    ref: shiftyfix
```

tronron, 2023-01-30:
So you got it working? Thats really exciting. Thank you.
I'll try this tonight and report back...

tronron, 2023-01-31:
Works perfectly! Thanks to both you, and the original author! This is a great integration!

Quattrohead, 2023-02-01:
Shifty and Tron, could you please give me step by step instructions on how to set this up in HA/esphome etc.
I am not a software guy but a very good hardware guy. I have got as far as having an esp32 junctec-wifi
ESPHome in my panel but no idea what to do with the 3 files here.

shiftybugger, 2023-02-02:
> Shifty and Tron, could you please give me step by step instructions on how to set this up in HA/esphome etc. I am not a software guy but a very good hardware guy. I have got as far as having an esp32 junctec-wifi ESPHome in my panel but no idea what to do with the 3 files here.

You don't need to do anything with the files in this repo. Just make sure that your ESPHome yaml for this device has the external component defined (as in my last post). When you hit 'install' in ESPHome, it will grab these components automatically and compile/link then into the firmware. After that you just integrate into HA the same as other ESPHome devices.

Let me know if you need more detail.

Quattrohead, 2023-02-02:
Sorry I have no idea where ESPhome yaml is.
I am looking around in the file editor and see a load of trash from my previous attempts to do this and a file called esphome-web-922e9e.yaml.json which is my esp32 device that is up and available on my network.
Maybe we can take this to email or PM.

shiftybugger, 2023-02-03:
> Sorry I have no idea where ESPhome yaml is. I am looking around in the file editor and see a load of trash from my previous attempts to do this and a file called esphome-web-922e9e.yaml.json which is my esp32 device that is up and available on my network. Maybe we can take this to email or PM.

I'm fairly new to esphome and HA myself but I'm happy to help you figure it out. Leave me an email and I'll drop you a line, rather than me try to explain stuff here.

tfyoung, 2023-02-06:
I've updated "main" to work again. Sorry for leaving it broken so long, I had to dismantle my battery system to renovate the room it was in and it was all in pieces. That's also taken up all my time. But it's done now and system is mostly back together :)

The proposed fix was broken due to the way checksuming was done, the previous code expected it to throw when it reached the end. I changed to checksuming to use a version of getval that returns an optional so it can still parse without knowing how many items to expect, but kept the main code as is with the longjmp fix.

Let me know if there are any issues with the new code.

closed 2023-02-06 by tfyoung