tfyoung/esphome-junctek_kgf #9: New sensors

Lukylic, 2023-02-18:
Hello, 
I would like to had the others sensors but I faild to do so... :-( 
I started first with ampHourRemaining I did update senso.py/ junctek_kgf.cpp/ junctek_kgf.h but there is mistake someware.... I cannot found were..... May you help me? 
[components.zip](https://github.com/tfyoung/esphome-junctek_kgf/files/10775026/components.zip)
[image]

tfyoung, 2023-02-28:
Can you make it as a pull request or share a git repo that has the diff so I can easily see what you've added?

Just by looking at the error though, I think you need to define `CONF_AMPHOUR_REMAINING` in sensors.py.

e.g. add ```CONF_AMPHOUR_REMAINING = "amphours_remaining". This is needed for config types which can't be pulled from esphome. The esphome const.py showing the included ones is at https://github.com/esphome/esphome/blob/dev/esphome/const.py If there is something sensible to describe the variable in that list, then you import it, if nothing makes sense then you need to declare it locally yourself.

E.g. In my other module for an air conditioner, I defined 
https://github.com/tfyoung/esphome-MHIHeatPump/blob/main/components/MHIHeatPump/sensor.py `CONF_OUTDOOR_TEMPERATURE = "outdoor_temperature"`

Lukylic, 2023-03-04:
Hello, thanks for your answer. I do not know what you mean by "a pull request or share a git repo". so I highlight on yellow w
![juncteck_kgf_h](https://us
[image]
er-images.githubusercontent.com/76538705/222923987-d4253300-c8bb-47d5-aaa9-5411e4fe1fba.jpg)
hat i did modify on you original Files. and i post it on this message. 
[image]

Lukylic, 2023-03-04:
[image]