tfyoung/esphome-junctek_kgf #10: how to make connections

niurz, 2023-02-19:
hi, thank you for the work you are doing.
I've been trying to use this method for several days but I can't read the data coming from the shunt. can you kindly tell us how to connect the shunt: to which of the two ports: the one of the display or the other?
the order of the cables to the rs-485 converter? which gpio on the esp32?
I tried all possible combinations but I can't read any data.
please help me
A thousand thanks

Quattrohead, 2023-02-22:
To read the data, you only need to connect GND and A of the RS485 port to your ESP OR **TTL to USB** device. Yes it is not RS485 so much as basic TTL with text output. I am able to read the output with a TTL converter into windows running YAT (Yet Another Terminal)
This is what you should see, cross reference to the manual to see what it all means.

:R50=1,<LF>:r50=1,208,5232,5710,923007,24276928,132091777,7900119,137,0,0,0,973,2154,
:R51=1,<LF>:r51=1,142,0,0,0,0,0,100,0,0,13000,100,100,100,0,0,1,

niurz, 2023-02-22:
I also knew this method but the problem is how do you import these data into HA? How do you create battery percentage, current and voltage sensor?

Quattrohead, 2023-02-22:
I dunno, I cannot for the life of me get this stuff to work. At least my devices seem to stick in ESPhome now with a static IP.

tfyoung, 2023-02-28:
I have an cheap rs485 ttl (something like https://www.aliexpress.com/item/1005001621746811.html) hooked up to the non display port. I forget the pin wiring, but I think the manual covers it.

I then have the other side of the rs485 hooked up the esp pins which i then define in the yaml as the uart. I also give it vcc and gnd on the uart side too.