tfyoung/esphome-junctek_kgf #18: error: call of overloaded 'millis()'

oli82, 2023-08-15:
Hi. 
I try to compile your code with esphome 2023.7.1 and get errors like that:

`src/esphome/components/junctek_kgf/junctek_kgf.cpp: In member function 'void JuncTekKHF::handle_settings(const char*)':
src/esphome/components/junctek_kgf/junctek_kgf.cpp:91:33: error: call of overloaded 'millis()' is ambiguous
   this->last_settings_ = millis();
                                 ^
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
                 from .piolibdeps/junctek-khf/ArduinoJson/src/ArduinoJson/Configuration.hpp:156,
                 from .piolibdeps/junctek-khf/ArduinoJson/src/ArduinoJson.hpp:7,
                 from .piolibdeps/junctek-khf/ArduinoJson/src/ArduinoJson.h:9,
                 from src/esphome/components/json/json_util.h:11,
                 from src/esphome/core/string_ref.h:10,
                 from src/esphome/core/entity_base.h:5,
                 from src/esphome/components/sensor/sensor.h:5,
                 from src/esphome/components/junctek_kgf/junctek_kgf.h:4,
                 from src/esphome/components/junctek_kgf/junctek_kgf.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:131:15: note: candidate: 'long unsigned int millis()'
 unsigned long millis();
               ^~~~~~
In file included from src/esphome/components/uart/uart.h:5,
                 from src/esphome/components/junctek_kgf/junctek_kgf.h:5,
                 from src/esphome/components/junctek_kgf/junctek_kgf.cpp:1:
src/esphome/core/hal.h:37:10: note: candidate: 'uint32_t esphome::millis()'
 uint32_t millis();
          ^~~~~~
src/esphome/components/junctek_kgf/junctek_kgf.cpp: In member function 'void JuncTekKHF::handle_status(const char*)':
src/esphome/components/junctek_kgf/junctek_kgf.cpp:133:30: error: call of overloaded 'millis()' is ambiguous
   this->last_stats_ = millis();
                              ^
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,`

Can you help me to get rid of this errors?

Thanks, Oli

krasa, 2023-10-04:
https://github.com/krasa/esphome-junctek_kgf/commit/21a3f79c674e50ef64574b5c34fade9ae848e18e

Arnix11, 2023-11-24:
Hi Oli,
I try to compile your code with esphome 2023.7.1 and get error like that:

INFO ESPHome 2023.11.4
INFO Reading configuration /config/esphome/esphome-web-daa12b.yaml...
Failed config

sensor.junctek_khf: [source /config/esphome/esphome-web-daa12b.yaml:40]

  
  [ah_total_used] is an invalid option for [sensor.junctek_khf]. Please check the indentation.
  ah_total_used: 
    name: junctek1_ah_total_used

Can you help me repair this error?
Thanks.
Arnold.

ldiegos, 2024-01-11:
Same error with millis() stuff... but following @krasa  I forked the project and add the esphome:: to my project and linked it to the esphome HA... is working fine now.

jbergler, 2024-03-06:
@krasa's fix does indeed work, but I can't seem to use the rest of the repo for some reason I haven't had time to debug.

@tfyoung I opened a PR that addresses the compilation issues without the other changes in @krasa's fork, if you'd be open to merging #23 it would make the component usable again.

fabikm, 2024-11-23:
MQTT Integration Example: Functional Code for Direct Implementation (Thanks to jbergler and @krasa)

# Enable MQTT
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  topic_prefix: "xxxxxx/xxxxxxx"

captive_portal:

# Publish sensor values to MQTT
time:
  - platform: sntp

interval:
  - interval: 5min
    then:
      - mqtt.publish_json:
          topic: "xxxxxx/xxxxx/xxxxxx"
          payload: |-
            root["voltage"] = id(battery_voltage).state;
            root["current"] = id(battery_current).state;
            root["battery_level"] = id(battery_level).state;
            root["temperature"] = id(battery_temperature).state;